InternRobotics/StreamVLN #2: Question on the effect of depth-assisted voxel pruning on inference time

Eku127, 2025-07-10:
Thanks a lot for sharing this great work and your open source plan!

I have a question regarding the use of depth information: as I understand, the depth is only used during inference time to assist voxel pruning for speeding up the model. However, I couldn't find detailed ablation or efficiency results on how much this pruning strategy accelerates the inference, especially in terms of single-frame latency.

Could you kindly share more details on the quantitative impact of depth-assisted pruning on inference efficiency?

Once again, thank you for the great contribution :)

kellyiss, 2025-07-10:
Thanks for your interest and kind words!

We included the token pruning ratio in our analysis, which reflects the percentage of tokens pruned. As the number of history tokens is reduced, the Transformer FLOPs also decrease. We agree that reporting FLOPs could be better.

Since VLN data is highly redundant, we believe this pruning strategy still has room for further improvement.
We also encourage more research on  pruning strategy to further boost efficiency

Eku127, 2025-07-10:
> I'm not sure if I understood your question correctly — we didn’t train the models of NavID or NaViLa. We only modified streamvln's textual representation of the actions to match their prompt formats.

[image]

Thanks! After rechecking the charts in NaVILA, NaVID, and UniNaVID, I think I’m starting to understand the setup.
If I’m correct:

- The entries without the dagger are results trained on R2R and RxR only.

- The entries with the dagger indicate results trained with augmented data, such as Room tours from YouTube for NaVILA, and Dagger data for StreamVLN.

- For StreamVLN, the “without dagger” results refer to performance after stage-1 training only, while the “with dagger” results are after mixture data training.

Eku127, 2025-07-10:
Appreciate the prompt and detailed response from the authors! Closing this issue.

closed 2025-07-10 by Eku127