InternRobotics/StreamVLN #7: Release StreamVLN training data on Hugging Face

NielsRogge, 2025-07-16:
Hi @mengwei0427 🤗

I'm Niels and work as part of the open-source team at Hugging Face. I discovered your work on Arxiv and was wondering whether you would like to submit it to [hf.co/papers](https://hf.co/papers) to improve its discoverability.If you are one of the authors, you can submit it at https://huggingface.co/papers/submit.

The paper page lets people discuss about your paper and lets them find artifacts about it (your dataset for instance),
you can also claim the paper as yours which will show up on your public profile at HF, add Github and project page URLs.

It's great to see that you've already released the StreamVLN model checkpoints on the Hugging Face Hub! This is fantastic for discoverability and usage.

I also noticed in your GitHub README's TODO list that you plan to "Release training data". Would you like to host this training data (which includes navigation-specific and general multi-modal data with DAgger augmentation) on https://huggingface.co/datasets when it's ready?
Hosting on Hugging Face will give you more visibility/enable better discoverability, and will also allow people to do:

```python
from datasets import load_dataset

dataset = load_dataset("your-hf-org-or-username/your-dataset")
```

If you're down, leaving a guide here: https://huggingface.co/docs/datasets/loading.
We also support Webdataset, useful for image/video datasets: https://huggingface.co/docs/datasets/en/loading#webdataset.

Besides that, there's the [dataset viewer](https://huggingface.co/docs/hub/en/datasets-viewer) which allows people to quickly explore the first few rows of the data in the browser.

After uploaded, we can also link the dataset to the paper page (read [here](https://huggingface.co/docs/hub/en/model-cards#linking-a-paper)) so people can discover your work.

Let me know if you're interested/need any guidance.

Kind regards,

Niels

bryce-wan, 2025-07-17:
Hi, we’ve already submitted our paper to [hf.co/papers/2507.05240](https://huggingface.co/papers/2507.05240) and uploaded our training data to [hf.co/datasets/cywan/StreamVLN-Trajectory-Data](https://huggingface.co/datasets/cywan/StreamVLN-Trajectory-Data). Let us know if anything needs adjustment on the Hub.

NielsRogge, 2025-07-19:
Thank you! That looks great