InternRobotics/StreamVLN #9: Evaluation results are significantly lower than reported in the paper

Eku127, 2025-07-18:
Hi, thank you for sharing the model weights. I evaluated the model `StreamVLN_Video_qwen_1_5_r2r_rxr_envdrop_scalevln` on the `val_unseen` split (`R2R_VLNCE_v1-3 preprocessed`) using 4 × L20 GPUs and followed your evaluation pipeline closely.

However, I observed that the results are significantly lower than those reported in your paper. Please see the attached screenshots for comparison:

[image] 
[image]

Interestingly, I ran the same evaluation setup for two other baselines, and their results are consistent with the paper. I was wondering if the released checkpoints for `StreamVLN_Video_qwen_1_5_r2r_rxr_envdrop_scalevln` are the same as those used to produce the final results reported in the paper?

I have attached my final results and looking forward to your advice and clarification—thank you!

[result.json](https://github.com/user-attachments/files/21310859/result.json)

zouc15, 2025-07-18:
[image]
I'm encountering the same issue

kellyiss, 2025-07-18:
Note: The current evaluation script does not include voxel pruning. The success rate (SR ≈ 55.7%) reported in Table 5 of our paper excludes voxel pruning and is the result that should be reproduced

zouc15, 2025-07-18:
> 1. Either updating the model config or modify the evaluation code can fix the bug.
> 2. We've confirmed that v1-3 gives more favorable initial headings, so training there likely produces better models. The performance gap should not be large, but we agree it’s better to evaluate all methods consistently on v1-3. Since our pipeline involves two-stage training and DAgger re-collection, updating all results will take some time. For now, note that RXR is not affected by this gap — you should be able to reproduce our RXR results directly.
> 3. The voxel pruning code still needs some cleanup before release — we’ll share it once it’s ready.

@kellyiss Thank you so much for your response! @Eku127 , I used the R2R-V1 dataset, which I think explains why my results are slightly better than yours.

Eku127, 2025-07-19:
I launched a task yesterday using the modified version of the codebase, running on the  `R2R-v1-preprocessed` dataset, and encountered a strange but familiar issue. It seems that the agent gets confused at a certain point in the environment and doesn’t proceed correctly.

I came across this related discussion: https://github.com/OpenRobotLab/StreamVLN/issues/3#issuecomment-3060439084, and i think following the instructions here will be a temporary fix.

At the moment, I haven’t had the chance to dig deeply into the code, so I’d appreciate any insights into what might be causing this behavior.

```
[22:07:52.235085] 256 You are an autonomous navigation assistant. Your task is to Turn around and exit the door into the workout room. Walk to the end of the first machine and turn left. Walk forward until you exit through the arch way into the living room
. Stand so you are in the living room even with the right side of the archway .  Devise an action sequence to follow the instruction using the four actions: TURN LEFT (←) or TURN RIGHT (→) by 15 degrees, MOVE FORWARD (↑) by 25 centimeters, or STOP. These a
re your historical observations <memory>.
[22:07:53.220130] <|im_start|>assistant
←→<|im_end|>
[22:07:53.220261] actions [2, 3]
[22:07:54.047105] <|im_start|>assistant
←→<|im_end|>
[22:07:54.047270] actions [2, 3]
[22:07:54.885067] <|im_start|>assistant
←→<|im_end|>
[22:07:54.885197] actions [2, 3]
[22:07:55.721102] <|im_start|>assistant
←→<|im_end|>
[22:07:55.721246] actions [2, 3]
[22:07:56.566053] <|im_start|>assistant
←→<|im_end|>
[22:07:56.566197] actions [2, 3]
[22:07:57.410130] <|im_start|>assistant
←→<|im_end|>
[22:07:57.410249] actions [2, 3]
[22:07:58.249022] <|im_start|>assistant
←→<|im_end|>
[22:07:58.249136] actions [2, 3]
[22:07:59.078242] <|im_start|>assistant
←→<|im_end|>
[22:07:59.078354] actions [2, 3]
[22:07:59.916063] <|im_start|>assistant
←→<|im_end|>
[22:07:59.916168] actions [2, 3]
[22:08:00.813957] <|im_start|>assistant
→→→→<|im_end|>
[22:08:00.814124] actions [3, 3, 3, 3]
[22:08:01.933141] <|im_start|>assistant
→→↑↑<|im_end|>
[22:08:01.933270] actions [3, 3, 1, 1]
[22:08:03.086283] <|im_start|>assistant
↑↑↑←<|im_end|>
[22:08:03.086390] actions [1, 1, 1, 2]
                                                                                                                                [22:08:04.227280] <|im_start|>assistant██████████████████████████████████████▌                   | 46/63 [20:38<06:00, 21.23s/it]
↑←←←<|im_end|>
[22:08:04.227389] actions [1, 2, 2, 2]
[22:08:04.717675] 290 You are an autonomous navigation assistant. Your task is to Turn around and exit the door into the workout room. Walk to the end of the first machine and turn left. Walk forward until you exit through the arch way into the living room. Stand so you are in the living room even with the right side of the archway .  Devise an action sequence to follow the instruction using the four actions: TURN LEFT (←) or TURN RIGHT (→) by 15 degrees, MOVE FORWARD (↑) by 25 centimeters, or STOP. These are your historical observations <memory>.
Traceback (most recent call last):
  File "/data/home//workspace/StreamVLN/streamvln/streamvln_eval.py", line 611, in <module>
    eval()
  File "/data/home//workspace/StreamVLN/streamvln/streamvln_eval.py", line 561, in eval
    evaluate(model, tokenizer, args)
  File "/data/home//workspace/StreamVLN/streamvln/streamvln_eval.py", line 580, in evaluate
    sucs, spls, oss, ones, ep_num = evaluator.eval_action(get_rank())
  File "/data/home//workspace/StreamVLN/streamvln/streamvln_eval.py", line 347, in eval_action
    outputs = self.model.generate(**input_dict, do_sample=False, num_beams=1, max_new_tokens=10000, use_cache=True, return_dict_in_generate=True, past_key_values=past_key_values)
  File "/data/home//miniconda3/envs/streamvln/lib/python3.9/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context
    return func(*args, **kwargs)
  File "/data/home//workspace/StreamVLN/streamvln/model/stream_video_vln.py", line 379, in generate
    ) = self.prepare_inputs_labels_for_multimodal(
  File "/data/home//workspace/StreamVLN/streamvln/model/stream_video_vln.py", line 224, in prepare_inputs_labels_for_multimodal
    cur_memory_feature = memory_features[batch_idx][cur_mem_id]
TypeError: 'NoneType' object is not subscriptable

```

kellyiss, 2025-07-19:
Outputting only two actions is not a bug. Although we explicitly train the model to predict 4-step actions, the output of large language models can be unstable sometimes. The model may occasionally fail to follow the expected pattern—especially when dealing with some hard cases.
This kind of behavior is not uncommon when finetuning large language models: even if we force them to follow a fixed template during training, they may still occasionally produce off-pattern or nonsensical outputs.

Eku127, 2025-07-19:
[image]

Thanks so much for your guidance! Based on your instructions, I’ve obtained results consistent with those in `Table 1` of the paper. Really appreciate the help!

If possible, could you update this issue with the results from training on` R2R v1-3 preprocessed` data as well as the corresponding checkpoint releases? If Stage 1 ckpt is also released that will be even greater 😆

closed 2025-07-21 by kellyiss