InternRobotics/StreamVLN #20: Question about the collection of navigation-specific data

LeonGoretzkatju, 2025-07-29:
Hi,

Thank you so much for your wonderful VLN work. I have some questions about the collection of navigation-specific data in your paper. In your paper, you write, "For navigation-specific data, we collect 450K video clips from R2R, R2R-EnvDrop, RxR, and ScaleVLN as expert training data, and 240K DAgger data samples as augmentation." However, when I tried to use the data from the HuggingFace link you provided, I only found R2R, RxR, and EnvDrop, with a lack of ScaleVLN data. Thus, I checked the original ScaleVLN project page and found that they only provide the navigation data in discrete VLN environment. Could you release the dataset or the dataset construction method of the ScaleVLN data part?

Thanks very much!

bryce-wan, 2025-07-29:
Hello, thank you for your interest in our work. Yes, we plan to release the ScaleVLN dataset annotations and episodes converted from discrete VLN soon. Stay tuned!

bryce-wan, 2025-07-30:
> > Hello, thank you for your interest in our work. Yes, we plan to release the ScaleVLN dataset annotations and episodes converted from discrete VLN soon. Stay tuned!
> 
> Hello,
> 
> Thank you so much for your reply. Could you please share the approximate timeline for open-sourcing datasets or dataset construction methods? I'm looking forward to seeing your released ScaleVLN-CE dataset.
> 
> Thanks a lot!

Hi, we have released the ScaleVLN episodes (CE) and annotations [here](https://huggingface.co/datasets/cywan/StreamVLN-Trajectory-Data/tree/main/ScaleVLN). If you encounter any issues during use, please feel free to contact us.

LeonGoretzkatju, 2025-07-30:
Hi,

Thank you for releasing the ScaleVLN-CE dataset. I tried the collection code following your instructions, and I didn't have any questions. 

Thank you!

closed 2025-07-30 by LeonGoretzkatju