InternRobotics/StreamVLN #21: Can you provide the code for generating annotations trajectories from R2R and RxR?

MuHeDing, 2025-07-30:
Hello autuhor. I am very interested in your work. Can you provide the code for generating annotations trajectories from R2R and RxR?
Like trajectory_data/EnvDrop/annotations.json, /R2R/annotations.json, and RxR/annotations.json.
I am very interested in the annotation code! ! !

bryce-wan, 2025-08-01:
Hello, here is an example:

```python
import habitat
import logging
import random
import json
import numpy as np
import argparse
import sys
import os
import torch

from omegaconf import OmegaConf
from PIL import Image

from habitat.tasks.nav.shortest_path_follower import ShortestPathFollower
from habitat_baselines.config.default import get_config as get_habitat_config

from habitat_extensions import measures

DATASET = "scalevln"
CONFIG_PATH = "path/to/habitat/config"
OUTPUT_PATH = "path/to/output/directory"

class CacheTrajectory:
    def __init__(self, dataset: str, config_path: str, output_path: str,):
        self.device = torch.device("cuda")
        self.dataset = dataset
        self.config_path = config_path
        self.output_path = output_path

        self.config = get_habitat_config(self.config_path)

    def config_env(self, scene=None):
        return habitat.Env(config=self.config)

    def generate(self):
        os.makedirs(os.path.join(self.output_path), exist_ok=True)
        env = self.config_env()

        scene_episode_dict = {}
        for episode in env.episodes:
            if episode.scene_id not in scene_episode_dict:
                scene_episode_dict[episode.scene_id] = []
            scene_episode_dict[episode.scene_id].append(episode)

        annotations = []
        for scene_id in sorted(scene_episode_dict.keys()):
            scan = scene_id.split("/")[-2]
            episodes = scene_episode_dict[scene_id]
            print(f"scene_id: {scene_id}, scan: {scan}")

            for episode in episodes:
                env.current_episode = episode
                agent = ShortestPathFollower(sim=env.sim, goal_radius=0.5, return_one_hot=False)

                instructions = episode.instruction.instruction_text
                trajectory_id = episode.trajectory_id
                scene_id = episode.scene_id.split('/')[-2]
                ep_id = int(episode.episode_id)
                ref_path = episode.reference_path

                observation = env.reset()

                # episode initialization
                rgb_list = []
                depth_list = []
                actions = [-1]
                next_waypoint_id = 1
                while not env.episode_over:
                    rgb = observation["rgb"]
                    rgb_list.append(rgb)
                    rgb_dir = os.path.join(self.output_path, "images", f"{scene_id}_{self.dataset}_{ep_id:06d}", "rgb_images")
                    os.makedirs(rgb_dir, exist_ok=True)
                    Image.fromarray(rgb).convert("RGB").save(os.path.join(rgb_dir, f"{len(rgb_list):03d}.jpg"))

                    next_action = agent.get_next_action(ref_path[next_waypoint_id])
                    
                    force_episode_over = False
                    while next_action == 0:
                        next_waypoint_id += 1
                        if next_waypoint_id == len(ref_path) - 1:
                            agent = ShortestPathFollower(sim=env.sim, goal_radius=0.25, return_one_hot=False)
                        if next_waypoint_id >= len(ref_path):
                            force_episode_over = True
                            break
                        next_action = agent.get_next_action(ref_path[next_waypoint_id])
                    
                    if force_episode_over:
                        break

                    observation = env.step(next_action)
                    actions.append(next_action)

                assert len(actions) == len(rgb_list), f"Actions length {len(actions)} does not match RGB frames length {len(rgb_list)}"    
                annotations.append({
                    "id": ep_id,
                    "video": os.path.join("images", f"{scene_id}_{self.dataset}_{ep_id:06d}"),
                    "instructions": instructions if isinstance(instructions, list) else [instructions],
                    "actions": actions,
                })
            with open(os.path.join(self.output_path, f"annotations.json"), "w") as f:
                json.dump(annotations, f, indent=4)

if __name__ == "__main__":
    parser = argparse.ArgumentParser()
    parser.add_argument("--dataset", type=str, default=DATASET)
    parser.add_argument("--config_path", type=str, default=CONFIG_PATH)
    parser.add_argument("--output_path", type=str, default=OUTPUT_PATH)
    args = parser.parse_args()

    runner = CacheTrajectory(
        dataset=args.dataset,
        config_path=args.config_path,
        output_path=args.output_path
    )
    runner.generate()
```

closed 2025-08-03 by kellyiss

MuHeDing, 2025-08-05:
> Hello, here is an example:
> 
> import habitat
> import logging
> import random
> import json
> import numpy as np
> import argparse
> import sys
> import os
> import torch
> 
> from omegaconf import OmegaConf
> from PIL import Image
> 
> from habitat.tasks.nav.shortest_path_follower import ShortestPathFollower
> from habitat_baselines.config.default import get_config as get_habitat_config
> 
> from habitat_extensions import measures
> 
> DATASET = "scalevln"
> CONFIG_PATH = "path/to/habitat/config"
> OUTPUT_PATH = "path/to/output/directory"
> 
> class CacheTrajectory:
>     def __init__(self, dataset: str, config_path: str, output_path: str,):
>         self.device = torch.device("cuda")
>         self.dataset = dataset
>         self.config_path = config_path
>         self.output_path = output_path
> 
>         self.config = get_habitat_config(self.config_path)
> 
>     def config_env(self, scene=None):
>         return habitat.Env(config=self.config)
> 
>     def generate(self):
>         os.makedirs(os.path.join(self.output_path), exist_ok=True)
>         env = self.config_env()
> 
>         scene_episode_dict = {}
>         for episode in env.episodes:
>             if episode.scene_id not in scene_episode_dict:
>                 scene_episode_dict[episode.scene_id] = []
>             scene_episode_dict[episode.scene_id].append(episode)
> 
>         annotations = []
>         for scene_id in sorted(scene_episode_dict.keys()):
>             scan = scene_id.split("/")[-2]
>             episodes = scene_episode_dict[scene_id]
>             print(f"scene_id: {scene_id}, scan: {scan}")
> 
>             for episode in episodes:
>                 env.current_episode = episode
>                 agent = ShortestPathFollower(sim=env.sim, goal_radius=0.5, return_one_hot=False)
> 
>                 instructions = episode.instruction.instruction_text
>                 trajectory_id = episode.trajectory_id
>                 scene_id = episode.scene_id.split('/')[-2]
>                 ep_id = int(episode.episode_id)
>                 ref_path = episode.reference_path
> 
>                 observation = env.reset()
> 
>                 # episode initialization
>                 rgb_list = []
>                 depth_list = []
>                 actions = [-1]
>                 next_waypoint_id = 1
>                 while not env.episode_over:
[... 142 lines cut ...]
>                 assert len(actions) == len(rgb_list), f"Actions length {len(actions)} does not match RGB frames length {len(rgb_list)}"    
>                 annotations.append({
>                     "id": ep_id,
>                     "video": os.path.join("images", f"{scene_id}_{self.dataset}_{ep_id:06d}"),
>                     "instructions": instructions if isinstance(instructions, list) else [instructions],
>                     "actions": actions,
>                 })
>             with open(os.path.join(self.output_path, f"annotations.json"), "w") as f:
>                 json.dump(annotations, f, indent=4)
> 
> if __name__ == "__main__":
>     parser = argparse.ArgumentParser()
>     parser.add_argument("--dataset", type=str, default=DATASET)
>     parser.add_argument("--config_path", type=str, default=CONFIG_PATH)
>     parser.add_argument("--output_path", type=str, default=OUTPUT_PATH)
>     args = parser.parse_args()
> 
>     runner = CacheTrajectory(
>         dataset=args.dataset,
>         config_path=args.config_path,
>         output_path=args.output_path
>     )
>     runner.generate()

Is the training set based on the three files R2R_VLNCE_v1-3_preprocessed/envdrop/envdrop.json.gz, RxR_VLNCE_v0/train_guide.json.gz, R2R_VLNCE_v1/train.json.gz as config_path?