InternRobotics/StreamVLN #23: Table 1 Consultation

newer79, 2025-08-01:
[image]
I would like to confirm whether the experiment within the red box was conducted by simultaneously mixing the Trajectory Data trained using R2R-CE and RxR-CE provided by Hugging Face, and then evaluating it on the two separate datasets?

missTL, 2025-08-04:
May I ask if the experiment within the red box utilized the ScaleVLN data provided by you?

missTL, 2025-08-04:
And are R2R-CE and RxR-CE provided by yourselves on Huggingface? Instead of the original complete R-CE and RxR-CE? Because I noticed that the R2R-CE and RxR-CE on Huggingface are not the same as those provided in VLN-CE.

closed 2025-08-04 by newer79