InternRobotics/StreamVLN #25: training on single node with A800-80G memory and CUDA out of memory.

ShengkaiWu, 2025-08-06:
With the following training script, I atampt to train the model on a single node with A800-80G memory. However, I encounter the error: CUDA out of memory.
streamvln_train_single_node.sh
`#!/bin/bash
#SBATCH --job-name=streamvln_single_node    # 任务名称
#SBATCH --gres=gpu:1                         # 默认1卡，实际由GPU_IDS控制
#SBATCH --time=96:00:00                      # 运行时间限制

# ==================== 显卡配置 ====================
# 可指定单卡或多卡，例如：
# export GPU_IDS=0             # 单卡
# export GPU_IDS=0,1,2,3       # 多卡
export GPU_IDS=3,5             # 默认使用第0号显卡

export CUDA_VISIBLE_DEVICES=${GPU_IDS}
NUM_GPUS=$(echo $GPU_IDS | tr ',' '
' | wc -l)

echo "使用显卡ID: ${GPU_IDS}"
echo "显卡数量: ${NUM_GPUS}"

# ==================== 路径与模型配置 ====================
VIDEO_FOLDER="data/trajectory_data/R2R","data/trajectory_data/RxR"
LLM_VERSION="Qwen/Qwen2-7B-Instruct"
VISION_MODEL_VERSION="google/siglip2-so400m-patch14-384"
PREV_STAGE_CHECKPOINT="lmms-lab/LLaVA-Video-7B-Qwen2"

# ==================== 训练参数配置 ====================
PROMPT_VERSION="qwen_1_5"
MID_RUN_NAME="StreamVLN_Video_${PROMPT_VERSION}_single_node_${GPU_IDS}"

echo "运行名称: ${MID_RUN_NAME}"
echo " checkpoint路径: ${PREV_STAGE_CHECKPOINT}"

# ==================== 单机训练命令 ====================
torchrun --nnodes=1 --nproc_per_node=${NUM_GPUS} \
    --master_addr=localhost --master_port=29500 streamvln/streamvln_train.py \
    --deepspeed scripts/zero2.json \
    --model_name_or_path ${PREV_STAGE_CHECKPOINT} \
    --version ${PROMPT_VERSION} \
    --video_folder ${VIDEO_FOLDER} \
    --group_by_task False \
    \
    --num_history 8 \
    --num_future_steps 4 \
    --num_frames 32 \
    --data_augmentation True \
    \
    --mm_tunable_parts="mm_vision_tower,mm_mlp_adapter,mm_language_model" \
    --vision_tower ${VISION_MODEL_VERSION} \
    --mm_projector_type mlp2x_gelu \
    --mm_vision_select_layer -2 \
    --mm_use_im_start_end False \
    --mm_use_im_patch_token False \
    --image_aspect_ratio anyres_max_9 \
    --image_grid_pinpoints  "(1x1),...,(6x6)" \
    --bf16 True \
    --run_name ${MID_RUN_NAME} \
    --output_dir checkpoints/${MID_RUN_NAME} \
    --num_train_epochs 1 \
    --per_device_train_batch_size 2 \
    --per_device_eval_batch_size 4 \
    --gradient_accumulation_steps 2 \
    --evaluation_strategy "no" \
    --save_strategy "epoch" \
    --save_total_limit 1 \
    --learning_rate 2e-5 \
    --mm_vision_tower_lr 5e-6 \
    --weight_decay 0. \
    --warmup_ratio 0.075 \
    --lr_scheduler_type "cosine_with_min_lr" \
[... 841 lines cut ...]
    return f(*args, **kwargs)
  File "/root/miniconda3/envs/streamvln/lib/python3.9/site-packages/torch/distributed/run.py", line 806, in main
    run(args)
  File "/root/miniconda3/envs/streamvln/lib/python3.9/site-packages/torch/distributed/run.py", line 797, in run
    elastic_launch(
  File "/root/miniconda3/envs/streamvln/lib/python3.9/site-packages/torch/distributed/launcher/api.py", line 134, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/root/miniconda3/envs/streamvln/lib/python3.9/site-packages/torch/distributed/launcher/api.py", line 264, in launch_agent
    raise ChildFailedError(
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
============================================================
streamvln/streamvln_train.py FAILED
------------------------------------------------------------
Failures:
  <NO_OTHER_FAILURES>
------------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2025-08-06_11:48:30
  host      : 71a7ed840277
  rank      : 0 (local_rank: 0)
  exitcode  : 1 (pid: 177925)
  error_file: <N/A>
  traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html
============================================================`

kellyiss, 2025-08-06:
Unfortunately, a single A800-80G GPU does not provide sufficient memory to support the full training run, which is why you're encountering a CUDA out of memory error. We recommend using at least 4×A100 80G GPUs.

ShengkaiWu, 2025-08-06:
The problem still exists. How many gpus do you use when you debug the code?

training scripts:
```bash
#!/bin/bash
#SBATCH --job-name=streamvln_single_node    # 任务名称
#SBATCH --gres=gpu:1                         # 默认1卡，实际由GPU_IDS控制
#SBATCH --time=96:00:00                      # 运行时间限制

# ==================== 显卡配置 ====================
# 可指定单卡或多卡，例如：
# export GPU_IDS=0             # 单卡
# export GPU_IDS=0,1,2,3       # 多卡
export GPU_IDS=0,1,2,3           # 默认使用第0号显卡

export CUDA_VISIBLE_DEVICES=${GPU_IDS}
NUM_GPUS=$(echo $GPU_IDS | tr ',' '
' | wc -l)

echo "使用显卡ID: ${GPU_IDS}"
echo "显卡数量: ${NUM_GPUS}"

# ==================== 路径与模型配置 ====================
VIDEO_FOLDER="data/trajectory_data/R2R","data/trajectory_data/RxR"
LLM_VERSION="Qwen/Qwen2-7B-Instruct"
VISION_MODEL_VERSION="google/siglip2-so400m-patch14-384"
PREV_STAGE_CHECKPOINT="lmms-lab/LLaVA-Video-7B-Qwen2"

# ==================== 训练参数配置 ====================
PROMPT_VERSION="qwen_1_5"
MID_RUN_NAME="StreamVLN_Video_${PROMPT_VERSION}_single_node_${GPU_IDS}"

echo "运行名称: ${MID_RUN_NAME}"
echo " checkpoint路径: ${PREV_STAGE_CHECKPOINT}"

# ==================== 单机训练命令 ====================
torchrun --nnodes=1 --nproc_per_node=${NUM_GPUS} \
    --master_addr=localhost --master_port=29500 streamvln/streamvln_train.py \
    --deepspeed scripts/zero2.json \
    --model_name_or_path ${PREV_STAGE_CHECKPOINT} \
    --version ${PROMPT_VERSION} \
    --video_folder ${VIDEO_FOLDER} \
    --group_by_task False \
    \
    --num_history 4 \
    --num_future_steps 2 \
    --num_frames 16 \
    --data_augmentation True \
    \
    --mm_tunable_parts="mm_vision_tower,mm_mlp_adapter,mm_language_model" \
    --vision_tower ${VISION_MODEL_VERSION} \
    --mm_projector_type mlp2x_gelu \
    --mm_vision_select_layer -2 \
    --mm_use_im_start_end False \
    --mm_use_im_patch_token False \
    --image_aspect_ratio anyres_max_9 \
    --image_grid_pinpoints  "(1x1),...,(6x6)" \
    --bf16 True \
    --run_name ${MID_RUN_NAME} \
    --output_dir checkpoints/${MID_RUN_NAME} \
    --num_train_epochs 1 \
    --per_device_train_batch_size 1 \
    --per_device_eval_batch_size 2 \
    --gradient_accumulation_steps 4 \
    --evaluation_strategy "no" \
    --save_strategy "epoch" \
    --save_total_limit 1 \
    --learning_rate 2e-5 \
    --mm_vision_tower_lr 5e-6 \
    --weight_decay 0. \
[... 880 lines cut ...]
  File "/root/miniconda3/envs/streamvln/lib/python3.9/site-packages/torch/distributed/run.py", line 797, in run
    elastic_launch(
  File "/root/miniconda3/envs/streamvln/lib/python3.9/site-packages/torch/distributed/launcher/api.py", line 134, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/root/miniconda3/envs/streamvln/lib/python3.9/site-packages/torch/distributed/launcher/api.py", line 264, in launch_agent
    raise ChildFailedError(
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
=======================================================
streamvln/streamvln_train.py FAILED
-------------------------------------------------------
Failures:
  <NO_OTHER_FAILURES>
-------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2025-08-06_14:43:20
  host      : 71a7ed840277
  rank      : 2 (local_rank: 2)
  exitcode  : -9 (pid: 193553)
  error_file: <N/A>
  traceback : Signal 9 (SIGKILL) received by PID 193553
=======================================================
```

[image]

kellyiss, 2025-08-06:
It seems that the issue you're encountering now is no longer related to out-of-memory errors, but rather to multi-GPU setup problems.  Maybe you can remove 
#SBATCH --job-name=streamvln_single_node    # 任务名称
#SBATCH --gres=gpu:1                         # 默认1卡，实际由GPU_IDS控制
#SBATCH --time=96:00:00                      # 运行时间限制

If you'd like to debug the code on a single GPU, that's totally fine. In that case, you can modify the argument to: --mm_tunable_parts "mm_mlp_adapter"

ShengkaiWu, 2025-08-06:
Thank you. I can now train the model by setting mm_tunable_parts="mm_vision_tower,mm_mlp_adapter" and per_device_train_batch_size=1, using 4 A800 GPUs. Here’s the GPU memory consumption:
[image]