InternRobotics/StreamVLN #32: The gt action annotations in R2R_VLNCE_v3

markinruc, 2025-08-07:
Dear author,

Great work!
I find that R2R_VLNCE_v3 provides annotated actions. Why doesn't streamVLN use these action annotations during training?

Thanks a lot!

bryce-wan, 2025-08-07:
Hi, thanks for the reminder. This is primarily due to the version of the dataset we initially used. During our training and evaluation, we used the v1 version of the [R2R-CE dataset](https://dl.fbaipublicfiles.com/habitat/data/datasets/vln/mp3d/r2r/v1/vln_r2r_mp3d_v1.zip) and we generated the action annotations ourselves. We’re not sure how these generated action annotations differ from those in v1_3.

Eku127, 2025-08-07:
> Hi, thanks for the reminder. This is primarily due to the version of the dataset we initially used. During our training and evaluation, we used the v1 version of the [R2R-CE [...]

So what is the behind reason to mix the r2r_v1 (from R2R) and r2r_v1-3 (from EnvDrop) for training? Why not just using the v1-3 version of the data❓

bryce-wan, 2025-08-08:
@Eku127 The discrepancies in data versions primarily stem from our different data sources. Initially, we obtained the R2R episodes from this official page https://github.com/facebookresearch/habitat-lab/blob/main/DATASETS.md, which led to variations between data versions. In fact, v1-3 provide higher-quality VLN episodes, and we are currently in the process of updating the data. Stay tuned.

markinruc, 2025-08-26:
@bryce-wan Hi, I noticed that you have updated v1-3 data on Huggingface. I wonder if training with v1-3 data can improve the performance. Can you share the newest experimental results?

Thanks a lot!

closed 2025-09-08 by markinruc