InternRobotics/StreamVLN #45: Loss spiking and performance drop when training with augmented data

MarSaKi, 2025-09-06:
Thank you for your generous open-source contribution. It's a fantastic project.

I've encountered some training instability during the fine-tuning process, particularly when using the **envdrop** and **scalevln** augmented data. The loss curve exhibits noticeable spikes, which I suspect might be impacting the model's performance. 

I ran the **r2r+rxr+envdrop** training twice with the exact same code and setup, but the results were significantly different, with a 20% variance in the Success Rate (SR).
Here are the details of the two training runs:

**First Training Run**

[image]

Performance:
{"sucs_all": 0.2267536704730832, "spls_all": 0.1994367287065086, "oss_all": 0.30886351277868407, "ones_all": 8.526830237804466, "length": 1839}

**Second Training Run**

[image]

Performance:
{"sucs_all": 0.4426318651441001, "spls_all": 0.39578010605422953, "oss_all": 0.5051658510059815, "ones_all": 6.444051254317853, "length": 1839}

As you can see, the difference in the final success rate is quite substantial.
Have you encountered a similar issue with training instability or significant performance variance between identical training runs? Any insights or suggestions on how to mitigate this would be greatly appreciated.
Thanks,

MuHeDing, 2025-09-07:
> Thank you for your generous open-source contribution. It's a fantastic project.
> 
> I've encountered some training instability during the fine-tuning process, particularly when using the **envdrop** and **scalevln** augmented data. The loss curve exhibits noticeable spikes, which I suspect might be impacting the model's performance.
> 
> I ran the **r2r+rxr+envdrop** training twice with the exact same code and setup, but the results were significantly different, with a 20% variance in the Success Rate (SR). Here are the details of the two training runs:
> 
> **First Training Run**
> 
> <img alt="Image" width="386" height="255" src="https://private-user-images.githubusercontent.com/24617881/486430238-8dc8d936-8551-4063-acfc-3d64a8d6682e.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTcyMzE4ODYsIm5iZiI6MTc1NzIzMTU4NiwicGF0aCI6Ii8yNDYxNzg4MS80ODY0MzAyMzgtOGRjOGQ5MzYtODU1MS00MDYzLWFjZmMtM2Q2NGE4ZDY2ODJlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTA3VDA3NTMwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQyYTZjMWNkYjllYjg2NzJmYjIyYmQ4ZDMzNTcwZjdiZmJlMGFhZmVhMDg3OTYwMTRhZjNhMjRjODg1NjFlODMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.EjQwKKoWnqLphoVUzh-un5gYT5aO3HrNzn2FxGlWKts">
> Performance: {"sucs_all": 0.2267536704730832, "spls_all": 0.1994367287065086, "oss_all": 0.30886351277868407, "ones_all": 8.526830237804466, "length": 1839}
> 
> **Second Training Run**
> 
> <img alt="Image" width="391" height="257" src="https://private-user-images.githubusercontent.com/24617881/486430282-3347db82-f063-404d-b168-a3da91b4ee21.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTcyMzE4ODYsIm5iZiI6MTc1NzIzMTU4NiwicGF0aCI6Ii8yNDYxNzg4MS80ODY0MzAyODItMzM0N2RiODItZjA2My00MDRkLWIxNjgtYTNkYTkxYjRlZTIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTA3VDA3NTMwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBlYWZjNGNlNjE3NzFlZDMxODMyODM4MTlmZjk0YjAyYWIzZTlhMGU4YmIzZGFjZDE1OGFkZWQ2OGJkNDJiYTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.S1LbR57EU_hTqVB1gjjlkeHkbkjEbGOw33xzVjdK7lA">
> Performance: {"sucs_all": 0.4426318651441001, "spls_all": 0.39578010605422953, "oss_all": 0.5051658510059815, "ones_all": 6.444051254317853, "length": 1839}
> 
> As you can see, the difference in the final success rate is quite substantial. Have you encountered a similar issue with training instability or significant performance variance between identical training runs? Any insights or suggestions on how to mitigate this would be greatly appreciated. Thanks,

Are the learning_rate 2e-5 and model_max_length 32768 in your code consistent? Have you added the lora layer?

MarSaKi, 2025-09-07:
> > Thank you for your generous open-source contribution. It's a fantastic project.
> > I've encountered some training instability during the fine-tuning process, particularly when using the **envdrop** and **scalevln** augmented data. The loss curve exhibits noticeable spikes, which I suspect might be impacting the model's performance.
> > I ran the **r2r+rxr+envdrop** training twice with the exact same code and setup, but the results were significantly different, with a 20% variance in the Success Rate (SR). Here are the details of the two training runs:
> > **First Training Run**
> > <img alt="Image" width="386" height="255" src="https://private-user-images.githubusercontent.com/24617881/486430238-8dc8d936-8551-4063-acfc-3d64a8d6682e.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTcyMzE4ODYsIm5iZiI6MTc1NzIzMTU4NiwicGF0aCI6Ii8yNDYxNzg4MS80ODY0MzAyMzgtOGRjOGQ5MzYtODU1MS00MDYzLWFjZmMtM2Q2NGE4ZDY2ODJlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTA3VDA3NTMwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQyYTZjMWNkYjllYjg2NzJmYjIyYmQ4ZDMzNTcwZjdiZmJlMGFhZmVhMDg3OTYwMTRhZjNhMjRjODg1NjFlODMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.EjQwKKoWnqLphoVUzh-un5gYT5aO3HrNzn2FxGlWKts">
> > Performance: {"sucs_all": 0.2267536704730832, "spls_all": 0.1994367287065086, "oss_all": 0.30886351277868407, "ones_all": 8.526830237804466, "length": 1839}
> > **Second Training Run**
> > <img alt="Image" width="391" height="257" src="https://private-user-images.githubusercontent.com/24617881/486430282-3347db82-f063-404d-b168-a3da91b4ee21.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTcyMzE4ODYsIm5iZiI6MTc1NzIzMTU4NiwicGF0aCI6Ii8yNDYxNzg4MS80ODY0MzAyODItMzM0N2RiODItZjA2My00MDRkLWIxNjgtYTNkYTkxYjRlZTIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTA3VDA3NTMwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBlYWZjNGNlNjE3NzFlZDMxODMyODM4MTlmZjk0YjAyYWIzZTlhMGU4YmIzZGFjZDE1OGFkZWQ2OGJkNDJiYTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.S1LbR57EU_hTqVB1gjjlkeHkbkjEbGOw33xzVjdK7lA">
> > Performance: {"sucs_all": 0.4426318651441001, "spls_all": 0.39578010605422953, "oss_all": 0.5051658510059815, "ones_all": 6.444051254317853, "length": 1839}
> > As you can see, the difference in the final success rate is quite substantial. Have you encountered a similar issue with training instability or significant performance variance between identical training runs? Any insights or suggestions on how to mitigate this would be greatly appreciated. Thanks,
> 
> Are the learning_rate 2e-5 and model_max_length 32768 in your code consistent? Have you added the lora layer? Finally, when testing, what is the loss?

Hi, it's all consistent with the origin config and no lora added.

soul10101, 2025-09-08:
May I ask if the same errors will occur when evaluating RxR? NE and SPL
[image]

kellyiss, 2025-09-09:
We have successfully reproduced the results on our side, and the outcomes are consistent across multiple runs. Other users have also been able to reproduce them. You may check the details in this issue for reference. https://github.com/InternRobotics/StreamVLN/issues/27